nodejs/hardware #1: Initial Agenda, Scope and Membership

mikeal, 2015-05-08:
Let's shoot for our first meeting next Friday.

Please comment here with ideas for the agenda, scope of this working group, and if you'd like to be involved as a member.

mikeal, 2015-05-08:
Also, I'll set up the first meeting and drive it but then I'm getting out of the way so I'll be looking for someone else to run this thing.

soldair, 2015-05-08:
I would like to be involved. I'm working on a number of projects that involve node/iojs on pis etc.

tmpvar, 2015-05-08:
sign me up!

emilyrose, 2015-05-08:
This is extremely relevant to my interests. Please count me in.

AnnaGerber, 2015-05-08:
Yes, I'd like to be involved

emilyrose, 2015-05-08:
Also pinging @frijol on this in case she is interested!

rvagg, 2015-05-08:
/cc @ajfisher @garrows

emilyrose, 2015-05-09:
As for the agenda and scope, here are a few things I'd like to propose:

* Discuss possibility of making serial a first-class citizen in core ("bless" node-serialport).
* Compiling materials/resources around the use of Node in hardware (for nodeschool, nodebots, etc.).
* Develop/maintain toolchain support for building/running Node on various embedded systems.

voodootikigod, 2015-05-09:
hi.

TheAlphaNerd, 2015-05-09:
*raises hand*

[image]

voodootikigod, 2015-05-09:
Serial should **not be** first class citizen, it tarnishes the original intent of node and IO.  Rather the necessary implementation points (SOCK_OPTS) should be exposed so like ruby-serialport, pyserial this can be done in just JavaScript and not have to cross the FFI boundary.

voodootikigod, 2015-05-09:
@mikeal I can run it but could you have picked a worse time :)

tmpvar, 2015-05-09:
/cc @rockbot @olizilla @kosamari @noopkat

emilyrose, 2015-05-09:
@voodootikigod that sounds like the kind of discussion I'm looking for! Maybe we take the in-depth details offline from this thread, but I'd like to explore this more.

voodootikigod, 2015-05-09:
@emilyrose yea def,, except someone has an anniversary right around the time we could do it in-person ;) Call anytime, you got my number right?

emilyrose, 2015-05-09:
@voodootikigod haha, yes yes. I do feel bad about that (just not nearly enough to miss an anniversary with my wife). I'm not sure if I have your number, so feel free to IRC/DM me if you want to chat.

I'd also like to open the conversation to others that may have input as well, so either way let's make sure we bring it up in subsequent meetings.

tmpvar, 2015-05-09:
I think @voodootikigod and @emilyrose have a great idea here - trying to remove the build step from getting started with node controlled hardware (over serial).  Maybe instead of having a private conversation we could pull it into a separate issue and begin collectively supporting an effort that will result in what we actually need.

I've kicked off an issue iojs/hardware#2 if you're interested in discussing it over there!

cjihrig, 2015-05-09:
I'm not sure how much I can offer here, but I'm interested in being involved.

emilyrose, 2015-05-09:
@tmpvar that's my main motivation for wanting to up the level of "out-of-the-box" support for serial connectivity. 

Thanks for starting the issue too; I will continue the conversation over there!

monteslu, 2015-05-09:
Count me in.

+1 to bundling serialport into iojs.  The build step was a major hurdle for many of the people participating in last year's NodeBots day events.  Things are a little better with the current method of delivering OS specific binaries, but this will always run the risk of getting out of sync.

An official serialport stream interface also puts more weight behind steering browser implementation direction.  Chrome is currently behind on a couple of features, and the others haven't shipped anything yet.

rwaldron, 2015-05-09:
Don't take this as my argument _against_ rolling Serialport into iojs, but as "things I'm concerned about". 

- If the iojs API is any different from `node-serialport`, there will be compatibility issues reminiscent of working in the DOM. 
- If node.js were to implement a blessed Serialport API, it would have to be identical to the iojs API or risk the same compatibility issues
- We've already created other "transports" that are modelled after the Serialport API, if the "blessed" implementation is different, then these will have to change and will potentially break their dependents

If we're talking about merging Serialport as-is, then none of the above matters.

mikeal, 2015-05-09:
@rwaldron what @voodootikigod is suggesting actually is that io.js expose SOCK_OPTS so that serial-port can be implemented without a build step or any native code and retain the same high level JS API.

bryan-m-hughes, 2015-05-10:
👋 I'm interested in participating.

One thing I would like to see discussed is non-Firmata based hardware systems where we run node/iojs directly on the device, such as the Raspberry Pi, BB Black, etc. Currently, [Raspi IO](https://github.com/bryan-m-hughes/raspi-io) (raspberry pi support for Johnny-Five) also requires a build step, but in this case it's to get efficient access to GPIO and PWM instead of the serial port. These are just a few initial thoughts I have so far.

What can we do to provide support in iojs for hardware peripherals other than the serial port? Can we do anything to make it easier to interface with C/C++ libraries, such as [Wiring Pi](http://wiringpi.com/) without a) a compile step or b) using FFI. FFI is a great tool, but as far as I am aware there is no way to asynchronously call a long-running C/C++ function.

Firmata based devices do have the advantage of having hard real time software in the mix, but running on device sometimes necessitates the use of C to avoid some of the unpredictability of timing due to GC and JavaScript in general. I realize this is probably _way_ outside the scope of _any_ iojs TC, but putting pressure on the V8 team to add asm.js support could go a long way towards alleviating this issue.

ajfisher, 2015-05-10:
Definitely interested in participating and agree totally with Rick that we
don't want to create a fork that unwinds a lot of work done on the libs
that are downstream deps.

I'm also interested in what this means for first class devices like BBB,
RPi etc but also the increasing capability of things like this Tessel 2 and
C.H.I.P that will be available later in the year. The potential for a full
JS stack on a very cheap board is fast approaching.

trevnorris, 2015-05-10:
Count me in.

garrows, 2015-05-10:
Count me in as well. 

If the decision does get made that the API changes (which isn't a huge deal if you write a shim too), take a look at Chrome Serial API implementation https://developer.chrome.com/apps/serial

voodootikigod, 2015-05-10:
ChromeSerial is based on node-serialport and still doesn't have parity with
node-sp and to my knowledge still in flux.

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

nodebotanist, 2015-05-13:
I'm interested in getting involved as well.

mikeal, 2015-05-13:
My schedule is really crazy right now but Chris called me and offered to step up and get this first meeting off the ground. I've added him to the G+ pages. @voodootikigod here's the docs on how to setup hangouts on air cause it is *not* simple :) https://github.com/iojs/io.js/wiki/Meetings:-Creating-a-Hangouts-On-Air-Event

voodootikigod, 2015-05-15:
Everyone, Mikeal originally scheduled this for today, happy to hold it today at 1PM EDT if that works for everyone. Sorry for the late notice, scrambling to get this in my calendar. If we do not do it today, I can't run it for another 3 weeks (Work, JSConf, Work) -- happy to let someone else take point on it.

Anyone disagree with 1PM EDT today?

nodebotanist, 2015-05-15:
I can listen in at that time, but I won't be super talkative.

nodebotanist, 2015-05-15:
I could run the meeting if you need to move the date, otherwise I'll be there.

voodootikigod, 2015-05-15:
https://plus.google.com/events/cra1gj1fedl24d3513n0v0bhagk

bryan-m-hughes, 2015-05-15:
Would it be at all possible to bump it back to 1:30? I'll be in another meeting from 1 to ~1:20ish.

voodootikigod, 2015-05-15:
Bryan just to confirm thats 1PM EDT not PDT. Still have the conflict
(timezones are the worst)

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

bryan-m-hughes, 2015-05-15:
Correct, 1PM EDT, 10AM PDT

voodootikigod, 2015-05-15:
Anyone object to moving it back 30 minutes?

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

voodootikigod, 2015-05-15:
For anyone wanting to talk: https://plus.google.com/hangouts/_/hoaevent/AP36tYfu1V9NHyK8SZx6RP7TS3GjQF8_p_Azz_ojv_Bgpk9eJndzpQ 

I hope that works.. I have no idea what I am doing.

soldair, 2015-05-15:
at least i caught the last minute =)

rwaldron, 2015-05-15:
Sorry I missed this, I was at a funeral.

cjihrig, 2015-05-15:
Is this group going to also encompass things like IoT?

voodootikigod, 2015-05-15:
Colin,

This would encompass IoT efforts.

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

Frijol, 2015-05-16:
cc @tcr @johnnyman727 @jia @kevinmehall @ekolker @nplus11 @harleykwyn @mrnice @sidwarkd little late to the conversation but all of you might be interested in this repo

RussTheAerialist, 2015-05-17:
I'm interested in being involved in this working group.

MrNice, 2015-05-17:
Compiling a series of JS hardware knowledge and learnings sounds like fun. Thank goodness Tessel2 uses nodejs (iojs). Count me in

johnnyman727, 2015-05-18:
I'm super tardy but I'd love to get involved as well.

joshmarinacci, 2015-05-21:
I would love to get involved. I've been doing a lot of Node on Pi work.

deadprogram, 2015-05-21:
Please also include me. Thanks!

voodootikigod, 2015-05-21:
The meeting was held already. As such I am closing this ticket.

closed 2015-05-21 by voodootikigod

tcr, 2015-05-22:
For anyone interested in watching the meeting, it was archived here: https://plus.google.com/events/cra1gj1fedl24d3513n0v0bhagk

boneskull, 2015-09-02:
Just saw this today.  If there's any way I can help or get involved, please let me know.

nebrius, 2015-09-02:
@boneskull we'd love to have you help! For now, keep an eye on #16 until we get our next meeting set up.