nodejs/hardware #3: Known Support Matrix

voodootikigod, 2015-05-09:
Much like we do in node-serialport it is vital to know a support matrix of known working and supported operating platforms to serial devices. Furthermore if this is a hardware group not just serial port should be represented include HID devices, Bluetooth (BLE) devices, and others. Together all these libraries (as well as many not listed) make up the hardware community using JS.

/cc @sandeepmistry @node-hid

reconbot, 2015-10-06:
The j5 platform support page is a good place to start. http://johnny-five.io/platform-support/#relationship:embedded

dtex, 2015-10-06:
Are these the only attributes that need to be tracked?

* node/npm version
* os
* device

rwaldron, 2015-10-06:
@reconbot @dtex that data is conveniently stored as JSON in a file https://github.com/bocoup/johnny-five.io/blob/master/src/platforms-plugins.json

nebrius, 2015-10-06:
Should we also track CPU architecture? I'm thinking of armv6 vs armv7 for the Raspberry Pi (the RPi 2 is armv7, but I feel like a lot of people still run armv6 builds of node).

rwaldron, 2015-10-06:
@nebrius makes sense

reconbot, 2016-07-11:
NodeJS itself doesn't make it easy to discover what ARM version you're on. I'm working on getting CI support for the missing platforms. Docker and QEMU has been a godsend.

I've got a pretty inclusive table on node serialport https://github.com/EmergingTechnologyAdvisors/node-serialport#platform-support

---
`serialport` supports and tests against the following platforms, architectures and node versions.

| Platform / Arch | Node v0.10.x | Node v0.12.x | Node v4.x | Node v5.x | Node v6.x |
|       ---       | --- | --- | --- | --- | --- |
| Linux / ia32    |  ☑  |  ☑  |  ☑  |  ☑  |  ☑  |
| Linux / x64     |  ☑  |  ☑  |  ☑  |  ☑  |  ☑  |
| Linux / ARM v6¹ |  ☐  |  ☐  |  ☐  |  ☐  |  ☐  |
| Linux / ARM v7¹ |  ☐  |  ☐  |  ☐  |  ☐  |  ☐  |
| Linux / ARM v8¹ |  ☐  |  ☐  |  ☐  |  ☐  |  ☐  |
| Linux / MIPSel¹ |  ☐  |  ☐  |  ☐  |  ☐  |  ☐  |
| Linux / PPC64¹  |  ☐  |  ☐  |  ☐  |  ☐  |  ☐  |
| Windows² / x86  |  ☑  |  ☑  |  ☑  |  ☑  |  ☑  |
| Windows² / x64  |  ☑  |  ☑  |  ☑  |  ☑  |  ☑  |
| OSX³ / x64      |  ☑  |  ☑  |  ☑  |  ☑  |  ☑  |

¹ ARM, MIPSel and PPC64¹ platforms are known to work but are not currently part of our test or build matrix. [#846](https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/846) ARM v4 and v5 was dropped from NodeJS after Node v0.10.

² Windows 7, 8, 10, and 10 IoT are supported but only Windows Server 2012 R2 is tested by our CI.

³ OSX 10.4 Tiger and above are supported but only 10.9.5 Mavericks with Xcode 6.1 is tested in our CI.

closed 2022-11-22 by Trott