nodejs/hardware #5: Adopt and maintain a Code of Conduct

bryan-m-hughes, 2015-05-15:
As discussed in https://github.com/nodejs/io.js/issues/1709, there is a baseline Code of Conduct, along with other governance docs, that each working group should adopt and maintain. Each group is free to modify as they see fit for their particular group.

The Code of Conduct is, in my opinion, the most critical of these and the one I propose we focus on first. NodeBots tends to have one of the better communities out there, and I want to make sure that this is reflected and furthered through the Hardware WG.

The current CoC for io.js is at https://github.com/nodejs/io.js/blob/v1.x/CONTRIBUTING.md#code-of-conduct. This CoC makes references to the [Citizen Code of Conduct](http://citizencodeofconduct.org/), which I am rather fond of and we adopted for [NodeBots SF](https://github.com/nodebots/sf/blob/master/coc.md).

nodebotanist, 2015-05-15:
+1 to all this.

voodootikigod, 2015-05-15:
:+1:

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

bryan-m-hughes, 2015-05-26:
I created two CoC proposals, modified specifically for the Hardware Working Group:
Based on the io.js CoC: https://gist.github.com/bryan-m-hughes/05c643fc3bc8e5034e08
Based directly on the Citizen Code of Conduct (from NodeBots SF technically): https://gist.github.com/bryan-m-hughes/3333e09e2bddaf6865df

The former puts us more in line with the rest of TCs/WGs, but I do kinda feel like it's thin on details. The later has lots of details, but isn't in line with the rest of the TCs/WGs.

I prefer the latter, but then again I might be a bit biased since I spearheaded NodeBots SF's adoption of it. What does everyone else think?

nodebotanist, 2015-05-26:
I prefer the latter-- let's blaze our own path on this :).

RussTheAerialist, 2015-05-26:
+1 what @nodebotanist said

bryan-m-hughes, 2015-05-26:
https://github.com/nodejs/hardware/pull/7

mikeal, 2015-05-27:
+1 on the latter, we'll should continue to learn from this and adopt better CoC's in the other WGs.

Resseguie, 2015-10-14:
Linking #21 because I think it should close this when it lands.

nebrius, 2015-10-14:
Agreed. There was talk previously on whether we should use the citizen CoC, but I feel like that should be handled by the diversity WG at this point in time.

closed 2015-10-14 by nebrius