nodejs/hardware #6: Central repository for NodeBots information

nodebotanist, 2015-05-15:
I've kind of dubbed this the NodeBots Ecosystem project

My call-to-action for the next meeting is to get together what I have on this project so far, and my ideas for what we can do to make it work and be a team project.

The fly-by is we need a site that shows the differences between the types and platforms of NodeBots, and guide new users toward a platform that meets their needs.

dtex, 2015-05-15:
This could be a page on http://nodebots.io

The http://johnny-five.io/platform-support/ page could be very helpful.

Perhaps we could talk to @rwaldron / Bocoup (do you know anyone who works there?) about sharing data.

mikeal, 2015-05-15:
Haha, Kassandra also works at Bocoup ;)

nodebotanist, 2015-05-15:
I think @dtex was joking. Either way, yes, I worked a little on the initial run of the new j5 site and planned to use lessons learned and some of the awesome features @rwaldron has implemented.

dtex, 2015-05-15:
I'm rarely serious

mikeal, 2015-05-15:
oh, haha! :)

Resseguie, 2015-05-16:
It would be nice to have it auto-generated from a central source like the johnny-five.io site.

nodebotanist, 2015-05-16:
That's why I've been building a database first. That way we can deploy new information quickly. More info tomorrow when I get some time to write.

mikeal, 2015-05-16:
The NodeSchool website has a lot of pages built from Google Spreadsheets, so that's an option.

closed 2022-11-22 by Trott