nodejs/hardware #8: 5/28 meeting, JSConf edition

bryan-m-hughes, 2015-05-27:
It's time for our next meeting. Many, if not most, of the people interested in participating are at JSConf right now, so let's do this in person!

Tomorrow (5/28) is the JavaScript IRL day, so let's host this meeting after we're done cleaning up, should be around 6:30 EDT or so.

Is there anyone who wants to participate but isn't at the conference?

bryan-m-hughes, 2015-05-27:
I propose the following high-level agenda:

- Introductions
- Follow up on individual action items from the previous meeting
- Discussion of new topics
- Creation and delegation of action items to individuals

dtex, 2015-05-27:
I'm in

RussTheAerialist, 2015-05-27:
I'm interested in participating, but not at the conference and will be without internet tomorrow.  Is it possible there will be a recording of this meeting also?

bryan-m-hughes, 2015-05-27:
@RussTheAerialist yes there will be a recording!

I'm thinking that we can do a Google Hangout if there are remote people who want to join, or we can just use iMovie or something else offline if it's just JSConf folks.

sandeepmistry, 2015-05-27:
I'm in!

bryan-m-hughes, 2015-05-27:
@sandeepmistry you're at JSConf, right? (I think we met last night)

Resseguie, 2015-05-27:
I'm in, too.

sandeepmistry, 2015-05-27:
@bryan-m-hughes yes, I'm at JSConf!

jacobrosenthal, 2015-05-27:
Ill be there.

jkleinsc, 2015-05-27:
Is this open to anyone at JSConf?

bryan-m-hughes, 2015-05-27:
@jkleinsc sure, stop on by!

bryan-m-hughes, 2015-05-28:
It doesn't look like there is anyone not at JSConf attending, so let's hold this by the pool! I'll update here once it's close to time for the meeting to start

nodebotanist, 2015-05-28:
I'll be there.

bryan-m-hughes, 2015-05-28:
Hey everyone, we're starting at 6:15 in the upper area by the pool. Feel free to grab a nice beach beverage and chill for a bit beforehand!

bryan-m-hughes, 2015-05-28:
We're at a table by right by the main door to the lobby. Starting very soon!

bryan-m-hughes, 2015-05-29:
Here's the recording from the event: http://youtu.be/kGvogHbndlw. Apologies if the audio isn't great, there was a lot of background noise.

reconbot, 2015-05-29:
It was great

dtex, 2015-06-01:
I dove in on my action item, but after reviewing the LIBUV code I'm not sure I can get up to speed quickly enough and I don't want us to possibly miss a release down the road. I'm going to keep ramping up on that code base, but I've asked @voodootikigod if he could contact them to keep this moving forward.

dtex, 2015-06-01:
Are we converting action items from this meeting to github issues?

dtex, 2015-06-01:
I did find this which says "LIBUV won't and can't do that":

https://github.com/libuv/libuv/issues/333

bryan-m-hughes, 2015-06-01:
@dtex 😥 welp...

We can talk about this more at the next meeting, but perhaps our course of action should be higher level, e.g. getting node-serialport into core.

Is anyone else going to be at NodeConf Adventure? I'll be there, and I'm hoping to get a chance to chat with some of the libuv folks.

BTW, I think that's a _great_ idea about opening issues for action items! Let's all do that.

dtex, 2015-06-01:
I'll be at NodeConf Adventure

jacobrosenthal, 2015-06-01:
That doesnt seem cut and dry to me. Wont and cant why.

bryan-m-hughes, 2015-06-01:
@jacobrosenthal I'm curious too

I created https://github.com/nodejs/hardware/issues/9 for my action item.

I'm going to go ahead and close this issue since we already had the meeting. Thanks everyone for pitching in!

closed 2015-06-01 by bryan-m-hughes

bryan-m-hughes, 2015-06-01:
BTW, so node-ffi _already_ has async capabilities! What's more, efforts are already underway to get it into node core: https://github.com/nodejs/io.js/labels/ffi

This far-fetched idea is starting to look really promising. I'm going to switch Raspi-IO over to using it to see how things go and document the process. https://github.com/nodejs/hardware/issues/11