nodejs/hardware #10: Expose ioctl in LIBUV

dtex, 2015-06-01:
If we could set rate, bit and parity for serial communications we would no longer need a compile step for Node-Serialport installations.

According to https://github.com/libuv/libuv/issues/333 this is not possible but we'd like to get a handle and why and explore alternatives.

fivdi, 2015-06-01:
[ioctl](http://man7.org/linux/man-pages/man2/ioctl.2.html) is a bit of a [God function](http://en.wikipedia.org/wiki/God_object). It's second parameter is more or less a device dependent blob that makes a ton of things possible, so not wanting to expose it in libuv is understandable. Also, it's a Linux/Unix function, so what should be used on Windows?

Exposing ioctl may be asking for too much. Perhaps it would be better to propose a concrete set of functions related to serial ports in libuv, for example, functions that allow the following properties to be set for a file descriptor:

* baudrate
* character size
* parity
* stop bits
* raw mode
* canonical mode
* ...

bryan-m-hughes, 2015-06-01:
I know we kinda sorta rejected this earlier, but maybe we should revisit integrating the node-serialport apis into node core?

fivdi, 2015-06-01:
If we look at the functionality provided by node today compared to 2009 when node-serialport was originally implemented, perhaps a lot of the functionality that's currently implemented in node-serialport is now available in node.

For example, could something like the following JavaScript code be used to create two streams, one stream for serial port receive and one for serial port send?

```
  rxstream = fs.createReadStream('/dev/ttyAMA0', {
    highWaterMark: <some number>,
    encoding: <some encoding>,
    flags: <some flags>
  });
  txstream = fs.createWriteStream('/dev/ttyAMA0', {
    highWaterMark: <some number>,
    encoding: <some encoding>,
    flags: <some flags>
  });

  rxstream.once('open', function (rxfd) {
    // rxfd available here, so call new libuv functions to set baudrate, parity, etc...
  });
  txstream.once('open', function (txfd) {

  });
```

These two streams could then be turned into a duplex stream with something like [duplexify](https://www.npmjs.com/package/duplexify). At the end of the day, there's going to be more to it that this of course, but would this not be a large portion of what node-serialport is today?

bryan-m-hughes, 2015-06-01:
@fivdi that approach has been researched in the past by @voodootikigod, and the tl;dr is that it just doesn't provide enough of what we need.

fivdi, 2015-06-01:
Can you tell me what it doesn't provide or link to an explanation of what it doesn't provide?

bryan-m-hughes, 2015-06-01:
ioctl

fivdi, 2015-06-01:
The sample code above doesn't provide ioctl, but it's not supposed to. Instead, it call calls currently non-existent functions from libuv in the rxstream `open` event handler:

```
  rxstream.once('open', function (rxfd) {
    // rxfd available here, so call new libuv functions to set baudrate, parity, etc...
  });
```

These new libuv functions would be used to set the baudrate, parity, character size, etc, ...

bryan-m-hughes, 2015-06-01:
Ah I'm sorry, I see what you're getting at now, I missed the "new methods to set baudrate, parity, etc" bit. People are always saying "why don't you just use TTY", so it's a bit of a sore spot.

voodootikigod, 2015-06-02:
Brian and Bryan,

Let me come up with the list of what we need added (sorry for delay was
kinda tied up with JSConf).

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

dtex, 2015-06-12:
This came up at NodeConf today and @mikeal mentioned that @saghul may already have it on his radar. Tagging both to make sure no work is being duplicated.

saghul, 2015-06-13:
This is the PR you can monitor. Help in reviewing is more than welcome!
https://github.com/libuv/libuv/pull/379

dtex, 2015-07-17:
This is tightly related to issue #2

nebrius, 2015-10-14:
Small update from the libuv PR listed above. That PR was closed and a new one opened at https://github.com/libuv/libuv/pull/484

dtex, 2015-11-16:
Hey gang, a new milestone has been added to libuv for the forthcoming 1.8.0 release and it does not include our favorite PR. I've added a [comment](https://github.com/libuv/libuv/issues/617#issuecomment-157089751) requesting its inclusion but if y'all can think of additional, compelling arguments in favor of it, I don't think that would hurt to add them.

dtex, 2015-11-17:
That may be part of it but @saghul said he's got some new notes to hand off to the requester and that it's unlikely this will land in 1.8.0

saghul, 2015-11-17:
Yes, my current plan is to slim the idea even more, and make it even more flexible (specially on the Unix side). Will update the libuv issue soon.

grahamehorner, 2018-02-06:
ping any update ?

saghul, 2018-02-07:
Not really. Alas this has stalled.

mk-pmb, 2019-12-03:
@dtex

> If we could set rate, bit and parity for serial communications we would no longer need a compile step for Node-Serialport installations.

Which OSes can we target? If it's only linux, it would probably be more secure to set said params with a `child_process` running `stty`, as that would expose a lot less of the more dangerous stuff. Also since it identidies devices by filename, permissions can be more fine-grained than superuser yes/no.

@nebrius

> People are always saying "why don't you just use TTY", so it's a bit of a sore spot.

Not sure whether you meant to specifically include `stty`; however, when people suggest stuff often, you'll get a better chance by making kind of an FAQ about what suggestions you've already tested and what they're lacking.

@grahamehorner

Would you mind sharing what you plan to use ioctl for?

greggwon, 2020-09-30:
Interaction with i2c and other GPIO pins has similar issues.  It's really important to manage all of these blocking moments in one way that is uniformly consistent.  epoll(2) and other blocking requests for the next event should be possible to use because applications have done this for years.

closed 2022-11-22 by Trott