nodejs/hardware #12: The GPL when using Node FFI

reconbot, 2015-06-01:
Placeholder for research. Currently it looks like you can't use non operating system GPL'd functions without distributing yourself under the GPL.

bryan-m-hughes, 2015-06-01:
Thanks for opening this. Something else I had a thought on: I wonder if "distribution" has an effect. I seem to recall that whether or not you redistribute the code can play a role.

So what I'm thinking is: let's say that we aren't _distributing_ any GPL code, even though we are dynamically calling it. Perhaps we could install these third party libraries by downloading them on the fly during installation, or perhaps wrap them up in an NPM package that's listed as a dependency (and thus also not distributed).

Can you find out if that changes the legality?

mikeal, 2015-06-02:
GPLv2's language is specific to "distributing." Using is not distributing.

AGPL has language to specifically including using and not distributing.

I'm not sure about the GPLv3.

reconbot, 2016-07-22:
I'm closing this as I think we're ok here.

closed 2016-07-22 by reconbot