nodejs/hardware #9: Prototype async node-ffi

bryan-m-hughes, 2015-06-01:
[node-ffi](https://www.npmjs.com/package/node-ffi) holds a lot of promise for eliminating the need for native code for modules, but there is one aspect that makes it less than ideal: everything is synchronous.

The C/C++ way of doing async programming is typically to spin up a background thread, but we don't have threading in JS. We could use something like [cluster](https://nodejs.org/api/cluster.html), but that approach isn't as efficient as threading.

So let's prototype what an node-ffi would look like with async capabilities baked in.

TooTallNate, 2015-06-01:
For the record, node-ffi _does_ expose an `.async()` function on every ForeignFunction instance, which invokes the function on the libuv thread pool. See https://github.com/node-ffi/node-ffi/blob/master/lib/_foreign_function.js#L65-L118

I'm also in the process of moving the core, low-level pieces of ref/node-ffi into io.js. It's a slow, tedious process since there's a lot to cover, but see https://github.com/nodejs/io.js/pull/1750, https://github.com/nodejs/io.js/pull/1759 and https://github.com/nodejs/io.js/pull/1762 (and a 4th for libffi itself next) if you want to follow the progress of that (basically the [`ffi`](https://github.com/nodejs/io.js/labels/ffi) label in the issue tracker).

bryan-m-hughes, 2015-06-01:
That's amazing @TooTallNate! I seriously had no idea there was an ```.async()``` function. Guess that's what I get for filing this issue before digging through the code 😅

I think this could be a _huge_ win for NodeBots especially since a lot of people doing hardware are beginners, and native modules can be a big barrier to entry.

Thanks for all your work getting ffi into core, I'll certainly be keeping an eye on things. If there's anything the Hardware WG can do to help, let us know!

closed 2015-06-01 by bryan-m-hughes