nodejs/hardware #14: Next Meeting

voodootikigod, 2015-06-25:
When works for everyone?

dtex, 2015-06-25:
I'm flexible

nebrius, 2015-06-25:
I'm pretty open next week, except for lunch on Monday and Tuesday (PDT). I'll be at CascadiaJS the following week (July 7th-11th) so my schedule will be erratic then.

reconbot, 2015-06-25:
Friday and Monday don't work for me

nebrius, 2015-07-09:
Picking up on this, what is everyone's availability this week and next week?

I can take the reins on this meeting since @voodootikigod is unavailable.

@dtex, @monteslu, and I are at Cascadia this week. I'm not opposed to having it this week, what do you two think?

dtex, 2015-07-09:
It's fine with me. Maybe Friday after talks and before dinner, say 5:30pm?

davidfekke, 2015-07-09:
Would that be 5:30 PM EDT?

nebrius, 2015-07-09:
That would be 5:30pm PDT/8:30pm EDT. Sounds good to me.

nebrius, 2015-07-09:
Does 5:30pm PDT/8:30pm EDT tomorrow, Friday July 9th work for everyone else?

cc @rwaldron, @nodebotanist, @emilyrose, @AnnaGerber, @Frijol, @ajfisher, @TheAlphaNerd, @trevnorris, @soldair, @johnnyman727. Anyone I missed?

nodebotanist, 2015-07-09:
Works for me

—Kas

Kassandra Perch
nodebotani.st

soldair, 2015-07-09:
yeah

TheAlphaNerd, 2015-07-09:
I can be there in person... I see two others from this thread next to me on the floor :P

nebrius, 2015-07-09:
Noooo, you can't see me! I'm a ninja >_> <_<

AnnaGerber, 2015-07-10:
Works for me

rvagg, 2015-07-10:
Will this be recorded and/or broadcast? I wouldn't mind tuning in to try and understand the exact purpose of this WG and what overlap (if any) there may be with the Build WG and also where the Build WG may be able to offer assistance.

Resseguie, 2015-07-10:
I'm out Friday but don't let that stop you. Looks like you've got a majority. I'll tune in to the recording.

nebrius, 2015-07-10:
@rvagg, yes it will be recorded/broadcast on Google Hangouts, and we'd love for you to join us! Details and links will be posted before the meeting tomorrow.

ajfisher, 2015-07-10:
Cool - I may have to grab the next one. I'm in China at the moment for
JSConf so I think this is on right about the time I'm presenting

johnnyman727, 2015-07-10:
I can't make it but hope to be at the next one!

rockbot, 2015-07-10:
I'm gonna crash this.

just... because.

nebrius, 2015-07-10:
@rockbot Yissss!

nebrius, 2015-07-10:
Hey everyone, just a reminder that our meeting starts in 1 hour! If you're at Cascadia, we're going to meet outside the main ballroom close to registration. If you're remote, I'll drop the Google hangout links here in about 30 minutes.

nodebotanist, 2015-07-10:
Unforseen circumstances mean I will not make it after all.

—Kas

Kassandra Perch
nodebotani.st

nebrius, 2015-07-11:
Hey everyone, here's the links:

YouTube stream link (for people who just want to listen): http://www.youtube.com/watch?v=ksNmqTN5LTU

Hangouts link (for people who also want to talk): https://plus.google.com/hangouts/_/hoaevent/AP36tYdQQVI4XtB2q3HKn6XjtN-GHndWBElbxRYkDeMidriS_STFEg

If you're at Cascadia, we're now meeting in the Green Room, which is by the right side of the stage area

rockbot, 2015-07-11:
the (youtube) stream is... not best :-/

Fishrock123, 2015-07-15:
Following up from this meeting, who is active here? I noticed the current @nodejs/hardware team definitely does not cover the actual people involved, and would like to rectify that.

nebrius, 2015-07-15:
Good call @Fishrock123. I'm active.

I'm going to attempt to make a list of candidates: @monteslu, @dtex, @AnnaGerber, @rwaldron, @soldair, @ajfisher, @Resseguie, @nodebotanist, @Frijol, @johnnyman727, @emilyrose, @TheAlphaNerd. Apologies to anyone I left off, and if I did, please do not take this to mean you're not included!

rwaldron, 2015-07-15:
Yo! 

@mattpodwysocki is working on our request to have MS manage a node-serialport test box. That's all I have to report.

Fishrock123, 2015-07-15:
Invitations sent to those mentioned people to join the @nodejs/hardware team, write (+issues) access to this repo , with @nebrius as a team maintainer (can add people to the team, etc). Let me know if you folks need anything else!

nebrius, 2015-07-15:
Thanks @Fishrock123, greatly appreciated!

emilyrose, 2015-07-17:
how can I help? I'd love to attend a meeting.

nebrius, 2015-07-17:
@emilyrose awesome, we'd love your help! I would watch this repo to see when we host the next meeting.

Perhaps you could also watch the recordings of the last meeting to see where we're at and drop a note on any thoughts you have? The tl;dr is our primary focus is on easing the pain of creating and distributing native modules. There is some work underway to get functionality in libuv/node core that will allow node-serialport to be all JavaScript, and we're looking at some other possibilities too.

nebrius, 2015-09-01:
Whoops, forgot to close this issue. Thanks everyone who attended

closed 2015-09-01 by nebrius