nodejs/hardware #16: Regular meetings?

nebrius, 2015-08-28:
Our meetings have been pretty sporadic since the group started. This is at least partly my fault, and I hope we can fix that!

At the Node summit recently, it was suggested we create a regularly scheduled meeting to help develop a cadence. What does everything think?

If we set up a regularly scheduled meeting, what times work best for everyone, keeping timezone differences in mind?

dtex, 2015-08-28:
I am available any time/day except Mondays and Fridays. My attendance is less important than some other on the team, so if it needs to be on one of those days I'll deal.

nebrius, 2015-09-01:
I'm pretty flexible on days/times. What does everyone think about making this monthly?

cc'ing various people who might be interested: @rwaldron, @nodebotanist, @emilyrose, @Resseguie, @rockbot, @AnnaGerber, @Frijol, @ajfisher, @TheAlphaNerd, @trevnorris, @soldair, @johnnyman727, @Fishrock123, @rvagg, @sandeepmistry, @jacobrosenthal, @reconbot. Let me know if I forgot anyone!

jacobrosenthal, 2015-09-01:
Im in

Qard, 2015-09-01:
I have some interest in attending as a representative of the docs wg. I'd like to see docs on how to use node with hardware. Don't worry about my schedule though. :smile:

nebrius, 2015-09-01:
@Qard fantastic, we'd love to have you!

Can you all please let me know what timezone your in?

It sounds like people are pretty flexible, so maybe we can just propose some times and see what sticks

Qard, 2015-09-01:
I'm Vancouver, so same as SF. Don't worry about timing for me though. I'm
flexible. :)

nebrius, 2015-09-01:
cc @BrianGenisio

BrianGenisio, 2015-09-02:
Tuesdays, Thursdays, and Fridays (work hours EST) work best for me.

reconbot, 2015-09-02:
Est I'm flexible on dates and times. Who wants to make plans on Mondays and
Fridays though? Good days to keep clear.

nebrius, 2015-09-02:
Sounds like everyone is pretty flexible, so I'm just going to throw out a suggestion.

How does the second Tuesday of every month at 1PM PST (4 PM EST) sound? It's during work hours and not-lunch hours for everyone in the US, late evening in Europe, and at least not middle of the night for eastern Asia and Australia.

reconbot, 2015-09-02:
Wfm

ajfisher, 2015-09-03:
Yep that sounds good. Early here in AU but at least doable. Other option is
shift it by a couple of hours each month so the "terribleness" of TZ is
being spread around.

boneskull, 2015-09-03:
I'd like to attend if that's OK. Sep 8 @ 13:00 UTC-08:00, correct?

nebrius, 2015-09-03:
good idea @ajfisher.

@boneskull Please do attend. It's at 13:00 UTC-07:00 right now due to daylight savings time (sorry, I should have said PDT not PST).

BrianGenisio, 2015-09-03:
I'm in.

sandeepmistry, 2015-09-05:
I'm in

nodebotanist, 2015-09-05:
I will attend as many as I can with my erratic schedule.

nebrius, 2015-09-08:
Hey everyone, just a reminder that our next hardware meeting is today at 1PM PDT, 20:00 UTC.

I recommend we use this meeting as a change to regroup and get caught up on the current state of things. Does that sound good to everyone else?

TheAlphaNerd, 2015-09-08:
I will still be on the road, and Wil unfortunately miss the meeting.

will done one be taking notes?

nebrius, 2015-09-08:
Would someone like to volunteer to be note taker? In any case, I'll be recording the meeting and will post it online.

BrianGenisio, 2015-09-08:
@nebrius +1 for notes and/or recording it.  I'd like to try to get connected next month.

nebrius, 2015-09-08:
ETA 5 minutes. Here are the links for the meeting:

Event link: https://plus.google.com/events/c4tk9pvuhag30m879im5ddl46gc. This link is for anyone who wants to speak, although there are only 10 slots.

YouTube Link: http://www.youtube.com/watch?v=0JtLl1EZ1v4. This link is for anyone who wants to just listen in, and there are an unlimited number of slots AFAIK.

boneskull, 2015-09-08:
Hmm, it appears I need to join Google+ to join?

nebrius, 2015-09-08:
@boneskull I do believe that's a necessity, yes

jacobrosenthal, 2015-09-08:
Either Im bad at google or I dont think the g+ one is a joinable hangout

nebrius, 2015-09-08:
hmm...let me double check

nebrius, 2015-09-08:
try this link: https://plus.google.com/hangouts/_/hoaevent/AP36tYd6W8iUdfcIzM0wWv84QRRkYdzJMI1CIY2zyLsFV0Wu2PM2gg?hl=en&authuser=0

nebrius, 2015-09-08:
It seems the second link is working fine, so can everyone please use that link? Sorry for the confusion. We'll wait 5 more minutes for people to join

boneskull, 2015-09-08:
note: you don't actually need a Google+ account.

Qard, 2015-09-08:
I'll participate in #io.js on IRC, if anyone is around. My internet is extremely unstable right now. :disappointed:

Resseguie, 2015-09-08:
FYI I'm listening in on listen-only channel.

Qard, 2015-09-08:
@boneskull If you want stuff to help with, the docs working group wants some introductory guides on using node on embedded devices. We'd greatly appreciate the help. :smile_cat:

nebrius, 2015-09-08:
@Qard that's a great suggestion! @boneskull, if this sounds like something you're interested, head over to https://github.com/nodejs/docs/blob/master/CONTRIBUTING.md and check out how to contribute.

nebrius, 2015-09-08:
Thanks everyone who was able to join. The recording is live at https://www.youtube.com/watch?v=0JtLl1EZ1v4&feature=youtu.be.

I don't think that @jacobrosenthal's chat was included in the audio recordings, so apologies for that everyone. The tl;dr is that we don't have support for Node.js 4 yet in node-serialport due to needing to upgrade to NAN 3. There is some refactoring work that needs to be done, which the team has been hesitant to do until now. Given the timing in finding other solutions, it may be necessary to go ahead and bit the bullet. Is that a good summation Jacob?

Action items from this meeting were:
- Follow up on the progress of getting access to devices to test on with Microsoft.
    - @rwaldron, do you have any updates to share?
- Follow up on the progress of ioctl integration into libuv
    - The current PR for ioctl support in libuv is at https://github.com/libuv/libuv/pull/484
    - @dtex, you've been tracking this progress right? Do you have anything to add?

dtex, 2015-09-08:
Sorry I missed. I was listening in for part of it on YouTube and will watch to the rest tonight. I'm going to try and get a summary of the state of the pull request from George. There are a few things that weren't clear on the face of it that I think may be problems. I'll update y'all when he replies (and I'm sure it'll be on  libuv/libuv#484 as well).

fivdi, 2015-09-08:
@jacobrosenthal Just in case you haven't seen them, there are two pull requests for upgrading node-serialport to NAN 2, https://github.com/voodootikigod/node-serialport/pull/571 and https://github.com/voodootikigod/node-serialport/pull/566

nebrius, 2015-09-08:
thanks for the links @fivdi. On the off-chance, have you looked at what the breaking changes in NAN 3 are? I haven't had a chance to do so yet.

Qard, 2015-09-08:
As far as I'm aware, NAN 2 was designed to be compatible with both io.js 3.x _and_ node 4.x. There is no NAN 3 yet.

fivdi, 2015-09-08:
It's NAN 2, NAN 3 doesn't exist yet :) I've upgraded two modules from NAN 1.8 to 2.0 so far and although a lot of modifications were required, they were relatively easy to make. For example, here's [the commit that upgraded i2c-bus](https://github.com/fivdi/i2c-bus/commit/70d6d2f35ce3714b780f68ca9acf2e83d9a5495e).

nebrius, 2015-09-08:
d'oh, my mistake. I thought I saw some conversation on twitter about it, but I was probably just reading iojs 3 😅

fivdi, 2015-09-08:
What is a little annoying at the moment is the fact that the "recommended" images for boards like the BeagleBone Black and Rapsberry Pi come with gcc 4.6 pre-installed, but to compile addons for io.js 3.x and Node.js 4.x gcc 4.8 is required.

nebrius, 2015-09-08:
:(

Fishrock123, 2015-09-08:
> but to compile addons for io.js 3.x and Node.js 4.x gcc 4.8 is required.

I'm sorry. There is literally nothing we can do about this. V8 moved to using c++11 features which require the version bump. It's out of our hands. :/

boneskull, 2015-09-08:
@Qard @nebrius thanks for the good suggestion.  I like writing, even.

C/C++ makes me cold and clammy, so if the bulk of the work to be done here is in that arena, I'll look to assist elsewhere.

nebrius, 2015-09-08:
@boneskull don't worry about lack of C/C++ skills. TBQH the bulk of the coding work is things to be done by other people. The bulk of what we do is make sure that we have a good understanding of what the pain points are and getting the right people connected and talking so that these pain points can be resolved, or at least lessened.

ajfisher, 2015-09-08:
Sorry I missed. 6am Australian time so when the reminder can't through it
was middle of the night.

Can I request whoever is going to coordinate (seems like Bryan) create and
share a calendar that we can subscribe to which will then take into account
local time zones etc...

vielmetti, 2015-09-08:
One other issue with gcc 4.6 -> gcc 4.8 support needed is noted here - https://github.com/voodootikigod/node-serialport/pull/566#issuecomment-138393439 - projects using Travis CI have to incorporate a few extra steps to make sure that they build their tests with the right compiler.

noopkat, 2015-09-12:
didn't know about this thread, sorry. I'm still interesting in participating in this WG.

nebrius, 2015-09-14:
@noopkat awesome, we'd love to have you! We don't have any formal membership or anything like that for this WG. I'd just watch this repo and keep an eye on issues filed.

noopkat, 2015-09-14:
ah thank you so much @nebrius, will do!

nebrius, 2015-10-05:
So I've been thinking. Our hangouts that we hold tend to be poorly attended and are basically a quick overview of what's going on. I don't feel like they have been very constructive in moving things forward.

We have some really productive discussions in GitHub issues though, so that tells me that maybe we should embrace that.

So I'm proposing a new structure for this working group: instead of doing a live recorded meeting, which is tricky with people in so many time zones, let's have a once a month "meeting in an issue" where we can all give a status update on everything.

What does everyone think about that?

We could also look in to setting up a Gitter chat, like we did for Johnny-Five, if we want something a little more synchronous than just GitHub.

noopkat, 2015-10-05:
@nebrius I like this idea a lot. I would love a Gitter channel too if possible

dtex, 2015-10-05:
I'm a big fan of Gitter and would be happy to see the conversations there.

nebrius, 2015-10-05:
Done and done! https://gitter.im/nodejs/hardware

nebrius, 2015-10-05:
I added a number of people to the room that were mentioned in this thread. I'm sorry if I forgot anyone! Feel free to add yourself or invite others. Also, feel free to leave the room if you want.

emilyrose, 2015-10-08:
I guess I'm going to start using gitter!

nebrius, 2015-10-20:
I think we've gotten a good flow regarding meetings (or the lack thereof now), so I'm closing this issue

closed 2015-10-20 by nebrius