nodejs/hardware #19: Chromium's hardware chops

joshperry, 2015-10-02:
Chromium has some pretty serious cross-platform hardware chops in its [web application platform](https://developer.chrome.com/apps/api_index), including support USB, bluetooth, bluetooth sockets, serial ports, bluetooth Low Energy, Scanners, HID, power management, printers, audio, etc.

They're even in the process of porting these libs to run natively on [mobile via Cordova].(https://github.com/MobileChromeApps/mobile-chrome-apps/blob/master/docs/APIsAndLibraries.md).

Has anyone looked at the possibility of leveraging the Chromium team's work to quickly bring some of this functionality to node?

I've written some pretty hardware intense apps (one is directly driving a fleet of USB-connected label printers) using these libraries and it's been a pretty decent experience that doesn't change between OS platforms and does not require futzing with native modules.

After working on one-such app, I thought it would be great to play with nw.js or Electron to set the app up as a standalone package. After about an hour just trying to get node-serialport to properly build and integrate, I had had enough. It was obvious pretty quickly that support for binary modules in these projects is brittle at best.

noopkat, 2015-10-02:
@joshperry Nice! There are a few of these projects already in npm I believe. There is a Chrome serialport, and a usblib version as well as perhaps others that I haven't looked into yet. The one catch with this is - do they work with Chromium specifically? I say this because in order to access these API's, you must be using them within a Chrome app itself, and not via client or server-side javascript code unfortunately.

Let me know if I'm talking about the wrong thing and misunderstood :)

PS. Totally feel your pain with electron wrt building native modules. I've abandoned doing that for now and use nw.js for that kinda stuff which is a little more straightforward once you figure out the tricks :joy:

noopkat, 2015-10-02:
here they are sorry 
https://github.com/jacobrosenthal/browser-usb
https://github.com/jacobrosenthal/chrome-serialport

joshperry, 2015-10-02:
@noopkat Ah, very interesting. If I understand correctly, these are basically the opposite of what I was thinking. :)

I was wondering if anyone had looked at the native implementation of the hardware APIs in Chromium to see if they'd be easy to pull out and repurpose for use in a node native module, or core if deemed appropriate.

noopkat, 2015-10-02:
@joshperry gotcha! Not that I'm aware of. I know that I was a little sad to see that it was limited to apps only, vs. asking the user for permission. 

Core would be a very difficult discussion IMO, something we have tabled; as it would require a lot of ownership and work from lots of people and would possibly stall their rapid releases. It's definitely something worth keeping in our thoughts.

As far as your other suggestion of repurposing - that would be awesome if it provides advantages over the current native modules. The Chrome stuff is way over my head so I'm not entirely sure if it's possible to do. If that's your area (sounds like you've done some cool stuff!!) so feel free to tackle this one head on :grin:

monteslu, 2015-10-02:
@joshperry I'd personally rather we stick to node's APIs and implement them on top of chrome's where possible.  At least until things leave the chrome. namespace and become official specs implemented in multiple browsers.

For bluetooth LE, I've been working on shimming noble on top of navigator.bluetooth here: https://github.com/sandeepmistry/noble/pull/238

joshperry, 2015-10-02:
@monteslu I agree 100%. My only goal in this group is to get good code for talking to hardware in the hands of node users, with a consistent, familiar, and well documented interface between all of us implementors. 

As a group we should be setting the standards for the interfaces that node users are learning and coding against. I just thought maybe we could steal some of their native code to fill out support in missing areas, beneath our own namespaces with our own API structures.

Making those interfaces available in the browser through a shim is a great idea for getting the most out of a user's code!

joshperry, 2015-10-02:
@noopkat I would definitely not be a fan of being tied to the core release cycles! I think keeping everything as modules is the best idea. I just want to make sure that we are getting help from core where appropriate (like the uv device patch they're working on getting out).

And I want to do more cool things! Interesting projects keep the talented people around and attract others.

I'm glad Javascript is cool again!😎

monteslu, 2021-03-03:
Resurrecting this old thread :)

WebHID, WebNFC, & Web Serial are all in stable as of chrome 89

WebBLE & WebUSB have been in stable for a while now and these are all open standards.  Would be awesome to have these in node.

I don't think there's much functionality left in the chrome namespace that node doesn't already have or isn't an open standard.

closed 2022-11-22 by Trott