nodejs/hardware #22: Branding

joshperry, 2015-10-27:
Had a few spare cycles and thought it'd be fun to prototype some branding specific to the node hardware WG.

Any other ideas/prototypes for branding, graphics, or whatnot?

[image]

[image]

I'm not a graphic artist but I did stay at a Holiday Inn Express last night.

reconbot, 2015-10-28:
hahaha, I love it!

Isn't there some movement to change the node logo?

nebrius, 2015-10-29:
Are we going to have another https://github.com/nodejs/node/issues/37 thread? Oh god I hope so :D

I like the play on the logo with the hardware connecting to it. What do you think about maybe including something more low level, maybe an LED or something, to highlight how Node.js is good for low and high level hardware-like things?

joshperry, 2015-10-29:
I think a thread like that would be awesome! Glad you guys like it, thought I'd kick something off.

@nebrius Yeah, I think an LED would be cool, and having an animated version where the LED flashed on and off occasionally, like the nodebots logo's eyes.

joshperry, 2015-10-29:
[image]

rvagg, 2015-10-30:
I want stickers of this one made up really small so I can put them alongside [image] and others on all my hardware!

nebrius, 2015-10-30:
Love the blinking LEDs, and yes, I would put those stickers everywhere too :D

Qard, 2015-10-30:
The important question is: does the tech exist to make the LEDs on the stickers blink? :open_mouth:

noopkat, 2015-10-30:
ok so we need to print a limited edition run of a paper circuit version where you can glue SMT leds in place and apply a battery :joy:

ekryski, 2015-12-05:
@joshperry that is legit. #nailedit

davidfekke, 2015-12-05:
I love that logo Josh.

dtex, 2015-12-05:
Sorry, accidental click