nodejs/hardware #24: Still active interest?

jandrieu, 2016-05-06:
Howdy.

It's been a while since there's been much conversation on the issues here. Is this still an active work group?  Has the conversation moved elsewhere?

@dshaw pointed me here at the Node.js Interactive conference in Portland (last December), but I've been swamped since then, so I know how it can get...

nebrius, 2016-05-07:
There's plenty of active interest in this group, but unfortunately what we're lacking in is people to actually _do_ things 😅. As a result, this WG is effectively on hiatus at the moment.

There's been interest in getting rid of the compile step in serialport to make the installation experience better for a lot of hardware libraries, but that progress has stalled due to lack of people with the libuv experience to get the PR landed.

On the enterprise embedded support front, no one has stepped up yet to try and get things done (which I saw you commented on already at https://github.com/nodejs/hardware/issues/18#issuecomment-217584930).

If you're interested in contributing time, that could go a long way towards getting things going again.

jandrieu, 2016-05-08:
I'm keen on using node with low-cost boards as an IoT brain, with or without various wireless connectivity (BLE, 6LowPAN, wi-fi).  That means CHIP, Raspery Pi (pick a flavor), Pine 64, BeagleBone, etc., as well as custom boards made using SOCs (System on Chip) or other multi-chip single-board systems.

Getting my BeagleBoneBlack running node last August was insane. I don't remember all the challenges, but it felt like one path after another leading to dead ends. Eventually I got it working, but *man* it should NOT have been so hard.

I'm just now trying to bring up my CHIP and will soon, .i.e., this quarter, will be trying the RaspPiZero and Pine64.

I'm not sure yet how to contribute, but I'll chime in here and see what might show up.

nebrius, 2016-05-09:
That's good feedback, thanks @jandrieu. Reading your post, I think the most valuable thing you can, if you have time, is to open an issue on this repo and document what issues you do run into for getting Node up and running on these devices. We can then direct it to the appropriate place, which could be the build WG, various Johnny-Five related repos, etc.

For the Raspberry Pi, I maintain support for the board for Johnny-Five over at https://github.com/nebrius/raspi-io/, and I have a write up for how to get Node and Johnny-Five working at https://github.com/nebrius/raspi-io/wiki/Getting-a-Raspberry-Pi-ready-for-NodeBots

rwaldron, 2016-05-09:
It's worth it to mention that the Tessel 2 comes with (currently) Node.js 4.4.3, but there is already a PR for the upgrading the system to Node.js 6. The update process for users is `t2 update` (assuming you've installed the t2-cli.

reconbot, 2016-08-06:
I think we should close this issue.

nebrius, 2016-08-08:
Agreed.

closed 2016-08-08 by nebrius