nodejs/hardware #27: Revival?

nodebotanist, 2018-12-19:
I'm willing to lead a revival of this working group. But I need to know if there's interest. (I also need to know what the right way to do this is).

Thinking we'd start in the new year if there's interest. Let me know what you think by commenting.

reconbot, 2018-12-20:
Yeah, that would be cool! The ecosystem has changed a bit since this was active and it would be neat to explore what that looks like.

mhdawson, 2018-12-20:
If there are enough interested people sounds great to me.  Looking at the README.md I wonder if making it a bit more generic (as nodebots talks about robots, but there are lots of other reasons to want to run JavaScript in small hardware devices)

dtex, 2018-12-20:
I'm on board, at least to discuss the current landscape of issues. We charted a path a few years ago but nothing really came to fruition and a lot has changed.

nodebotanist, 2018-12-20:
Okay then, looks like there's at least some interest.

@dtex , was the path chart documented somewhere?
@mhdawson, I'll start hashing out the README. Any ideas what you'd like to add?
@reconbot exactly, between Tessel and Microsoft MakeCode there's a lot of new projects to bring in.

dtex, 2018-12-20:
@nodebotanist Not sure if anyone took notes, but the big issues landed in issues here. The two big things I remember are:
* CI test setup for node commits to make sure things don't break on SBC's. Requires hardware and a place to put it so needs a sponsor.
* Landing support for iOCTL-like functionality in LIBUV so that serialport no longer requires binaries. I bet @reconbot has learned a lot about the feasability of that.

reconbot, 2018-12-20:
I learned that we need a stream like struct for serial ports in libuv as the file based ones wont work on windows as well as a way to get/set local state, and read remote state and receive events for changes on some of them.

mhdawson, 2018-12-20:
Maybe just change `The purpose of this team is to identify and remove barriers to entry for working with NodeBots. We are focused on barriers to entry that are related to Node.js and surrounding technologies, such as NPM.` to

`The purpose of this team is to identify and remove barriers to entry for working with JavaScript on smaller hardware devices (for example: NodeBots). We are focused on barriers to entry that are related to Node.js and surrounding technologies, such as NPM.`

nodebotanist, 2018-12-23:
All right, I'll get right on the changes proposed by @mhdawson, @dtex, and @reconbot. I'll try to schedule a meeting in the new year. Please reach out to anyone you think would interested in attending, and I'll create an issue for it so that folks can register interest in attending. 

Thank you, all!

nodebotanist, 2019-02-01:
I am still working on this-- will get a meeting proposal out soon, just onboarding with a new job :)

nebrius, 2019-02-04:
I’ll help out.

nodebotanist, 2019-02-07:
Please post your availability using the doodle above before tomorrow night (Feb 8) at 9PM CST. That's when I'll pick the most popular time and invites will go out.

Working on the Agenda here: https://docs.google.com/document/d/1RBJFKNDQ43G-PVOlNXyb6sySGarotvOkgbYllxWJFlo/edit?usp=sharing if you have a bullet point you'd like to add please drop me a note on this thread.

nodebotanist, 2019-02-11:
Tomorrow at 6PM CST (sorry for short notice) -- calendar link here https://doodle.com/poll/zuyw8zktkx35se8t

nodebotanist, 2019-02-12:
Hangout Link: https://meet.google.com/wnh-kaiu-beb?authuser=1

nodebotanist, 2019-02-13:
Hooray! Meeting successful! Minutes are in the google doc (I totally forgot to record it, super sorry!) https://docs.google.com/document/d/1RBJFKNDQ43G-PVOlNXyb6sySGarotvOkgbYllxWJFlo/edit?usp=sharing

Trott, 2019-02-13:
This doesn't have to be part of y'all's mission if you don't think it fits, but just expressing my own wish here: I'd love to see a hardware track at Node+JS Interactive. Or just more hardware/automation/robots/internet-of-things stuff going on at the community level at the conference and the collaborator summit. That's more of a tactical thing than a strategic thing, but it's also something super-concrete and in the not-too-distant future.

nebrius, 2019-02-13:
I _love_ that idea @Trott, and I'd be down to help make that happen! Do you know who the best point person on the Node+JS Interactive side of things to talk to about getting the ball rolling would be?

Trott, 2019-02-13:
> I _love_ that idea @Trott, and I'd be down to help make that happen! Do you know who the best point person on the Node+JS Interactive side of things to talk to about getting the ball rolling would be?

I don't. Maybe @ladyleet is doing the planning or knows who is?

I realize that the next collaborator summit won't be that one, though! It will be at the end of May in Berlin adjacent to JSConf EU. Maybe hook up with the folks talking in https://github.com/nodejs/summit/issues/135 to get more involved in that if anyone can/will go?

nebrius, 2019-02-13:
Thanks Rich. I will be at JSConf EU, FWIW, and I believe @nodebotanist will be too, so that seems doable.

mhdawson, 2019-02-13:
A comment on one of the questions in the minutes.  To be a working group the group needs to be formally chartered by the TSC.  The path most often followed is to start as a team, get things rolling do work etc.  At some point later somebody is surprised it's not a working group based on the work being done, it's proposed and then it gets chartered.  Day to day it does not make much of a difference if it is a team or a WG anyway so I'd start by focussing on getting work started.

Trott, 2019-02-13:
Make sure you know exactly what you're getting out of being chartered that you don't get out of being an _ad hoc_ team. In many cases, it makes sense to remain an _ad hoc_ team.

reconbot, 2019-02-13:
I’ve started researching the libuv proposal, my work on the web serial spec and node serial port are speaking to it. I think it’s going to be a scaled back stream struct that doesn’t do any configuration or opening (opposed to the currently suggested incomplete configuration solution). Should have a draft in a few days.

ladyleet, 2019-02-13:
@nebrius we will have a call for papers soon that @chowdhurian is prepping with Jory if you'd like to propose a session at the Collab Summit! :) you should follow details here: https://github.com/nodejs/summit/issues/135

ladyleet, 2019-02-13:
@nebrius same for the October event - we will have a CFP and you can follow the summit.

ladyleet, 2019-02-13:
PS @nodebotanist love your initiative!!!! <3

Trott, 2019-02-16:
Something I'd love to see (but might be out of scope?) is Android support. As far as I'm aware, the biggest blocker for officially supporting Android at this time is that we don't have a way to build and test on Android in CI.

reconbot, 2019-02-16:
I do know there are projects for iOS and Android that include node. One
project that uses it is https://www.manyver.se/

-Francis

mhdawson, 2019-02-20:
To avoid opening an new issue.  As an FYI discussion considering dropping support for ARM6 - https://github.com/nodejs/build/issues/1677#issuecomment-465326568

boneskull, 2019-02-20:
Just heard about this through nodejs/build#1677.  Am interested in helping out.  Please loop me in on any further doodles (email in profile)

nodebotanist, 2019-02-25:
Thanks for the heads up @mhdawson -- luckily y'all beat me to it with your concerns, so I will keep an eye on that issue. This is definitely an issue that we will want to champion, as it affects many hardware/Node.JS users.

@boneskull, welcome. We post meeting information in this repo, and we will attempt to reach out for the next meeting.

reconbot, 2019-03-15:
I got my draft going #29 needs review/feedback however as I've never proposed anything to a working group meeting I don't know the format/structure/completeness needed.

closed 2022-11-22 by Trott