nodejs/hardware #28: TC53 Representation

nodebotanist, 2019-02-25:
I'll be at the TC53 (embedded JS) meetings in ATX on Tues the 26th and Wed the 27th, I'll be a little late both days though due to medical appointments. 

If there's anything you'd like asked, please post here and I will convey any appropriate questions to the best of my ability.

Also, anyone else here going? @nebrius I think you mentioned you might make it.

nebrius, 2019-02-25:
@nodebotanist unfortunately I can't make it in person anymore. The timing didn't quite work, plus I'm getting over a cold now.

I will try and join remotely if I can, however looking at my schedule on those days, I think I'm going to have to bow out of this one. I'll try and make the next one!

dtex, 2019-02-25:
I’ll be there both days. Bryan will be joining us remotely on Tuesday. 

One thing to keep in mind when considering questions, is that TC-53 is focused on embedded systems running JS and node is not well suited for those environments as they have very limited resources.

However, I’m hoping that anything that comes out of TC-53 can also be applied to scenarios where node makes sense (SBC’s and remote devices), so I’m trying to make sure that is considered.

nebrius, 2019-02-25:
Just to clarify, I've got a lot of other meetings tomorrow, so I won't be able to join the TC-53 meeting till around noon PST(2pm CST), and will have to drop off by 3:30 PST (5:30 CST), so it'll unfortunately have to be brief.

closed 2022-11-22 by Trott